AndrewFromMelbourne/raspi2png #6: raspi2png fails to resize rotated capture

mihalop, 2015-05-29:
Under:

* latest Raspbian image (2015-05-05)
* have a rotate display - `display_rotate=1`
* display resolution: 1050x1680
* running raspi2png using: `./raspi2png --height 840`

A black screenshot is saved. I tried many combinations for height/width, but all failed to save the image (actual display resolution height/width worked).

AndrewFromMelbourne, 2015-05-29:
I will have a look at it. I am a little busy right now, may take me a couple of days to find time to try and reproduce it. Possibly an issue with the Raspberry Pi firmware.

AndrewFromMelbourne, 2015-06-05:
OK I can reproduce it. It is a firmware issue. I have a code sample that displays the issue. I will raise an issue on the Raspberry Pi firmware (soon).

AndrewFromMelbourne, 2015-06-06:
The issue has now been raised against the Raspberry Pi firmware. https://github.com/raspberrypi/firmware/issues/435

AndrewFromMelbourne, 2015-06-17:
There is a fix available in the latest firmware. You can either use rpi-update to retrieve the latest firmware or wait for the firmware to appear in Raspian (if that is distribution you are using).

mihalop, 2015-06-17:
Great Andrew, thanks!

closed 2015-06-17 by mihalop