AndrewFromMelbourne/raspi2png #8: raspi2png raspi2png: vc_dispmanx_snapshot() failed error

matttheorbiter, 2015-07-19:
I have installed and reinstalled this on my Pi 2. I can take one or two snapshots. Then it always fails with raspi2png: vc_dispmanx_snapshot() failed. If I reboot, sometimes it will let me take snapshots, sometimes not. If I delete the directory and reinstall the app, sometimes I can get it to do one or two. Then the error returns. Thanks for your help.

AndrewFromMelbourne, 2015-11-01:
Hi, sorry I completely missed this issue. Have you managed to resolve the problem? I would think that it may have been an issue with a particular version of the Raspberry Pi Firmware.

larryfant, 2016-10-12:
At least I have that problem when playing video with omxplayer. Without playing video screencaptures are ok.

AndrewFromMelbourne, 2016-10-12:
Thanks @larryfant I will see if I can reproduce it.

Jabroni, 2017-01-24:
Any resolution / update on the issues? Im experiencing same issue, what is wierd is that Im able to take screenshots of videos that have still images on it (using omxplayer), but if there is movement on the video, and I try to run raspi2png,  I get "raspi2png: vc_dispmanx_snapshot() failed "

AndrewFromMelbourne, 2017-01-24:
No, unfortunately I am pretty short of time these days. The error comes from the Raspberry Pi firmware rather than my code.

Jabroni, 2017-01-24:
Seems this is being caused cuz of the raspberry is running out of GPU memory. I added 
gpu_mem=96 to my raspberry config.txt and i was able to take the screenshot of the videos that had more motion on it

AndrewFromMelbourne, 2017-01-25:
Hi @Jabroni. Yes I should have thought of that. Glad you have found a solution/work around to your problem.

closed 2020-01-28 by AndrewFromMelbourne