AndrewFromMelbourne/raspi2png #9: raspi2png is too slow

k1-hedayati, 2015-11-01:
It almost takes 2 seconds to take a picture, is it possible to improve performance?

AndrewFromMelbourne, 2015-11-01:
No, not really.

AndrewFromMelbourne, 2015-11-02:
I have added a command line option to set the compression level of the PNG image. If you set a low compression level (default is 6), then the compression of the image should take less time. However the resulting PNG file will be larger and may take longer to save.

k1-hedayati, 2015-11-02:
Thank you, actually I want to convert screenshot to jpg for lower image size, because I want to transmit it over wire, so let's see if imagemagic can convert image faster or not

AndrewFromMelbourne, 2015-11-02:
Have a look at @dividuum's [screenshot](https://github.com/info-beamer/tools/tree/master/screenshot). It takes a snapshot and writes a JPG to stdout.

dividuum, 2015-11-03:
It should also be possible to use OMX to offload the jpeg encoding to the GPU. Since the CPU version works well enough for now I didn't look into that yet.

AndrewFromMelbourne, 2015-11-03:
@dividuum Yes, you could use OMX or MMAL to encode the jpeg on the GPU. On the Pi 2B it may not be worth the extra effort.

k1-hedayati, 2015-11-03:
@AndrewFromMelbourne thank you @dividuum's screenshot is fast and file size is small. Feel free to close issue.

closed 2015-11-03 by AndrewFromMelbourne