AndrewFromMelbourne/raspi2png #10: screen shot error

navaneetharaopy, 2015-12-01:
Board: raspberry pi - 2
OS:  Raspbian wheezy
Setup: Omxplayer with `--win` option running video and browser open at the back.

If i take screenshot  in this setup(mentioned above) i get following error
```
$ ./grabdisplay
tvservice-client: Failed to connect to TV service: -1
./grabdisplay: unable to get display information
```

I have this [link](https://github.com/raspberrypi/firmware/issues/373) which says "gencmd service is limited to 4 concurrent clients". Is there any solution for this?
like checking `bcm_host_init()` before loading

AndrewFromMelbourne, 2015-12-01:
I will have a look. No promises though.

navaneetharaopy, 2015-12-01:
Ok, Thanks

AndrewFromMelbourne, 2015-12-01:
Hi, unfortunately I have run out of ideas. It is the call to bcm_host_init() that fails, but if I don't call it then the call just fails later on. I believe it is a limitation of the DispmanX API.

navaneetharaopy, 2015-12-01:
Ok

navaneetharaopy, 2015-12-01:
can we check whether bcm_host_init is already available and  if not initialize by calling `bcm_host_init()`.

Just like header file defination 
`#ifndef SOME_HEADER_FILE_H`
`#define SOME_HEADER_FILE_H`

AndrewFromMelbourne, 2015-12-01:
Unfortunately no. I actually tried just not calling bcm_host_init(), but as I said above the code just fails later on.

closed 2020-01-28 by AndrewFromMelbourne