AndrewFromMelbourne/raspi2png #12: TV sercvice error on latest Raspbian

gkreidl, 2016-03-08:
tvservice-client: Failed to connect to TV service: -1
raspi2png: unable to open display 0

Using latest firmware on RPi3

AndrewFromMelbourne, 2016-03-08:
Thanks for the report. Unfortunately, I don't have a RPI3. Can you try rebuilding the executable and see if that helps?

gkreidl, 2016-03-08:
I compiled it again, but doesn't make a difference. I suppose it won't work on a RPi2 either after updating to the latest firmware (not using rpi-update, just the official release).

> Thanks for the report. Unfortunately, I don't have a RPI3. Can you try rebuilding the executable and see if that helps?
>

AndrewFromMelbourne, 2016-03-08:
Do you have the experimental OpenGL driver enabled?

gkreidl, 2016-03-08:
No

AndrewFromMelbourne, 2016-03-09:
I can't reproduce the problem on a Pi2. I had been on the NEXT branch of the Raspberry Pi, but re-installed the raspberrypi-bootloader package.

```
$ uname -a
Linux rpi1024b 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux
```

```
$ vcgencmd version
Feb 25 2016 14:25:47
Copyright (c) 2012 Broadcom
version dea971b793dd6cf89133ede5a8362eb77e4f4ade (clean) (release)
```

gkreidl, 2016-03-09:
Strangely enough it is working now after rebooting.
I had another problem with tvservice command a few days ago. Maybe it crashed in special circumstances and the problem is not related to raspi2png at all. I'll report if it happens again.

AndrewFromMelbourne, 2016-03-09:
I would be surprised if raspi2png stopped working and other programs that use the DispmanX interface still worked. It is possible that another program you are using is corrupting the GPU. The raspi2png program is very simple, most of the issues relate to issues with the firmware.

AndrewFromMelbourne, 2016-03-09:
Closing ... reopen if as necessary.

closed 2016-03-09 by AndrewFromMelbourne