AndrewFromMelbourne/raspi2png #26: Doesn't work on Raspbian Buster: libpng12 dependency issue

ZLevine, 2020-01-24:
I've been using this amazing library for years after setting it up [per this guide](https://howchoo.com/g/mgmym2fimzd/how-to-take-screenshots-on-the-raspberry-pi); however, I tried on a fresh install of Raspbian Buster and get the following error:

```
raspi2png: error while loading shared libraries: libpng12.so.0: cannot open shared object file: No such file or directory
```

It looks like raspi2png relies on libpng12, which isn't included in the later versions of Raspbian. Additionally, it can't be installed manually since there's no download link for [the armhf version on the package listing](https://packages.ubuntu.com/xenial/armhf/libpng12-0/download).

Any idea what we can do here to get it working? Thanks!

AndrewFromMelbourne, 2020-01-26:
Let me have a look

ZLevine, 2020-01-26:
Thanks! Let me know if you need help testing anything out.

AndrewFromMelbourne, 2020-01-28:
I have pushed a new binary built on Raspbian Buster. Please git it a try and let me know if there are any issues.

ZLevine, 2020-01-29:
Tested and working! Thanks a ton.

closed 2020-01-29 by ZLevine