AndrewFromMelbourne/raspi2png #33: raspi2png: unable to open display 0

GabeCoder, 2021-04-06:
Whenever I run ```raspi2png```, the output says, ```raspi2png: unable to open display 0```. Do you know how this issue can be solved? Is this a bug or a software problem?

susheelbanyal, 2021-12-09:
Is there any update on this issue? I am facing this issue on Pi4 with Bullsye. Please update.

closed 2021-12-09 by VertucioSoftware

AndrewFromMelbourne, 2021-12-28:
Hi, Sorry for the very long delay. I haven't been active with the Raspberry Pi for a long time. I don't know why exactly this issue is happening. It might be worth asking in the Raspberry Pi forums.

free5ty1e, 2025-01-27:
This is what I get as well on my raspberry pi zero 2 w running 64-bit bookworm (no lxde / windowed mode) as well.  Tried via SSH as well as directly on the pi via USB keyboard, does not make a difference.  Did anyone ever find out anything about this?

AndrewFromMelbourne, 2025-01-28:
Unfortunately, the API called DispmanX that is used by this program is no longer supported by the Raspberry Pi OS and will never work on new models like the Raspberry Pi 5. I should archive this and other projects that user DispmanX and update the README.

AndrewFromMelbourne, 2025-01-29:
In addition, you can get this working by by editing /boot/firmware/config.txt and changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d`. But note that this isn't officially supported in the Raspberry Pi OS and it could stop working at any time. Again this will only work for Raspberry Pi models up to and including the model b 4.

free5ty1e, 2025-01-29:
Thank you, that is good info - I didn't realize all that.  That will work for now :)