AndrewFromMelbourne/raspi2png #35: unable to open display 0

Botspot, 2022-02-10:
Raspberry Pi OS Bullseye now uses the Full KMS OpenGL driver by default, and not surprisingly, raspi2png doesn't work.
I use this tool often to take screenshots with the mouse pointer and of OpenGL overlays. No other tool can do that.

Raspi2png is in [the Pi-Apps app store](https://github.com/Botspot/pi-apps) and over one thousand people use it. As we continue to make the app store 100% compatible with Bullseye, we will be forced to remove your app if it cannot comply. Would you be willing to look into this again?

Botspot, 2022-02-10:
Here's the full `strace` log:
```
pi@raspberrypi:~ $ strace raspi2png
execve("/usr/bin/raspi2png", ["raspi2png"], 0xff9cc9c0 /* 59 vars */) = 0
brk(NULL)                               = 0x1fa9000
uname({sysname="Linux", nodename="raspberrypi", ...}) = 0
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xf7c4b000
close(3)                                = 0
readlink("/proc/self/exe", "/usr/bin/raspi2png", 4096) = 18
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v8l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xf7c49000
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf7c06000
mprotect(0xf7c0a000, 61440, PROT_NONE)  = 0
mmap2(0xf7c19000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0xf7c19000
close(3)                                = 0
munmap(0xf7c4b000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=121592, ...}) = 0
mmap2(NULL, 121592, PROT_READ, MAP_PRIVATE, 3, 0) = 0xf7be8000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libbcm_host.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\2446\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=67372, ...}) = 0
mmap2(NULL, 147120, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf7bc4000
mprotect(0xf7bd4000, 61440, PROT_NONE)  = 0
mmap2(0xf7be3000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xf000) = 0xf7be3000
mmap2(0xf7be5000, 11952, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xf7be5000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libpng16.so.16", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\200E\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=189752, ...}) = 0
mmap2(NULL, 253960, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf7b85000
mprotect(0xf7bb2000, 65536, PROT_NONE)  = 0
mmap2(0xf7bc2000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2d000) = 0xf7bc2000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libm.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\20\222\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=386572, ...}) = 0
mmap2(NULL, 450684, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf7b16000
mprotect(0xf7b74000, 61440, PROT_NONE)  = 0
mmap2(0xf7b83000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x5d000) = 0xf7b83000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\240\255\1\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=1321488, ...}) = 0
mmap2(NULL, 1390760, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf79c2000
mprotect(0xf7b01000, 61440, PROT_NONE)  = 0
mmap2(0xf7b10000, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x13e000) = 0xf7b10000
mmap2(0xf7b14000, 6312, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xf7b14000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libvchiq_arm.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0`\25\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=26064, ...}) = 0
mmap2(NULL, 92080, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf79ab000
mprotect(0xf79b1000, 61440, PROT_NONE)  = 0
mmap2(0xf79c0000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x5000) = 0xf79c0000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libvcos.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0t*\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=38752, ...}) = 0
mmap2(NULL, 105340, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf7991000
mprotect(0xf799a000, 61440, PROT_NONE)  = 0
mmap2(0xf79a9000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8000) = 0xf79a9000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
[... 73 lines cut ...]
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x2, 0x1c), 0xffc5d1a8) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0x4fb000d) = 0
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0x4fb100e) = 0
mmap2(NULL, 8392704, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS|MAP_STACK, -1, 0) = 0xf57fd000
mprotect(0xf57fe000, 8388608, PROT_READ|PROT_WRITE) = 0
[... cut ...]
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x8, 0x10), 0xffc5d294) = -1 EAGAIN (Resource temporarily unavailable)
futex(0xf7be7dd0, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, NULL, FUTEX_BITSET_MATCH_ANY) = 0
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x8, 0x10), 0xffc5d294) = 4
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0x4fac009) = 0
write(2, "raspi2png: unable to open displa"..., 36raspi2png: unable to open display 0
) = 36
exit_group(1)                           = ?
+++ exited with 1 +++
```

A2-Dev, 2023-07-07:
More of a workaround, but if you change the dtoverlay, then raspi2png will work again.
Stumbled upon this trying to troubleshoot another old program on the new raspbian version. (So feel free to ignore this if it's not helpful)

/boot/config.txt
Replace :
`dtoverlay=vc4-kms-v3d`
With :
`dtoverlay=vc4-fkms-v3d`