AndrewFromMelbourne/raspi2png #36: Kind Request: Tinker compile

2play, 2022-03-19:
I can see its using the bcm library for compiling/use on pi. Would you kindly consider add Asus Tinker?

2play, 2022-07-13:
Dear @AndrewFromMelbourne  thank you for still checking and replying on the subject. I found a tool that does all I need on rockchip boards, not bcm lib dependent and works on x11, framebuffer and kms.

closed 2022-07-13 by 2play

saper-2, 2022-07-14:
> There is a Universal Screenshot Tool in [fb2png](https://github.com/AndrewFromMelbourne/fb2png). The nice thing about rsapi2png is that it could capture GPU output. There is no longer support for DispmanX on the Raspberry Pi, so this tool is no longer of much use.

Hi, sorry for tapping into this topic 😄 
Is there any tool that can now (on Pi that use latest raspbian 11 (or whatever is called now 😉 )) tap into GPU output and "grab screenshot"?

AndrewFromMelbourne, 2022-07-15:
@saper-2 not that I know of. I have not been involved with Raspberry Pi's for a number of years. It would be best to ask this question in the Raspberry Pi forums.

2play, 2022-07-15:
@saper-2 get this and compile it (very fast) put the binary to your path or copy to /usr/bin and it will create a snapshot  of anything on raspberry. I dont recall is Ive tested on Bullseye version but if not i have another for you
the screenshot tool would make jpg out or anything you are on (desktop, framebuffer, kms)

saper-2, 2022-08-27:
@2play thanks for info, looks promising, but I have problem with linking:
```
pi@raspberrypi:~/tools/tools/screenshot $ make
cc -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -std=gnu99 -fgnu89-inline  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -O3 -Wno-psabi -s -L/opt/vc/lib/ -lbcm_host -lvcos -lvchiq_arm -lpthread -lrt -ljpeg screenshot.c -o screenshot
/usr/bin/ld: /tmp/ccak44ng.o: in function `main':
screenshot.c:(.text.startup+0x34): undefined reference to `bcm_host_init'
/usr/bin/ld: screenshot.c:(.text.startup+0x3c): undefined reference to `vc_dispmanx_display_open'
/usr/bin/ld: screenshot.c:(.text.startup+0x48): undefined reference to `vc_dispmanx_display_get_info'
/usr/bin/ld: screenshot.c:(.text.startup+0x64): undefined reference to `vc_dispmanx_resource_create'
/usr/bin/ld: screenshot.c:(.text.startup+0x78): undefined reference to `vc_dispmanx_snapshot'
/usr/bin/ld: screenshot.c:(.text.startup+0x94): undefined reference to `vc_dispmanx_rect_set'
/usr/bin/ld: screenshot.c:(.text.startup+0xc4): undefined reference to `vc_dispmanx_resource_read_data'
/usr/bin/ld: screenshot.c:(.text.startup+0xcc): undefined reference to `vc_dispmanx_resource_delete'
/usr/bin/ld: screenshot.c:(.text.startup+0xd8): undefined reference to `vc_dispmanx_display_close'
/usr/bin/ld: screenshot.c:(.text.startup+0xe8): undefined reference to `jpeg_std_error'
/usr/bin/ld: screenshot.c:(.text.startup+0x104): undefined reference to `jpeg_CreateCompress'
/usr/bin/ld: screenshot.c:(.text.startup+0x118): undefined reference to `jpeg_stdio_dest'
/usr/bin/ld: screenshot.c:(.text.startup+0x134): undefined reference to `jpeg_set_defaults'
/usr/bin/ld: screenshot.c:(.text.startup+0x144): undefined reference to `jpeg_set_quality'
/usr/bin/ld: screenshot.c:(.text.startup+0x150): undefined reference to `jpeg_start_compress'
/usr/bin/ld: screenshot.c:(.text.startup+0x45c): undefined reference to `jpeg_write_scanlines'
/usr/bin/ld: screenshot.c:(.text.startup+0x480): undefined reference to `jpeg_finish_compress'
/usr/bin/ld: screenshot.c:(.text.startup+0x488): undefined reference to `jpeg_destroy_compress'
collect2: error: ld returned 1 exit status
make: *** [Makefile:8: screenshot] Error 1
```
I did some google-fu and find somewhere that on arm64 might not everything compile properly (there is some issues with libraries or library paths).
I'll try try armhf (32bit) raspbian image too.

2play, 2022-08-31:
@saper-2 sorry for the delay
I had/ve this on my PlayBox but not tested on x64 setup.
On Rockchip x64 i use kmsgrab as the libs for any other solution is based on bcm libs
I now prefer later as its not bcm restricted